tscircuit/tscircuit-autorouter #66: Implement trace thickness as a parameter

seveibar, 2025-03-20:
You must have a major contribution prior to attempting this.

I think this issue is really hard and the bounty is likely to increase over time. There are a lot of implications of trace thickness for high density routing and probability calculation etc.

I'm not even really sure anyone should try to attempt to get trace thickness to go in a single PR, it probably will take a lot of effort.

We mainly only care about trace thickness multiples (which might help with the implementation). I.e. right now every trace is the industry standard data line thickness `0.15mm`, but we really only need to add support for 2x, 4x and 8x multiples, like `0.3mm`, `0.6mm`  and `1.2mm`

In many cases, thickness isn't really adequate to specify what the user wants. The user wants to make sure enough power can flow to particular chips. This can be done by creating more traces to the chip, or by routing thicker lines. Routing more traces is "autorouter friendly", we're more likely to succeed because we can weave the trace through obstacles. Thick traces are more "human friendly", it's easy to see what carries power.

In practice I believe having good trace adjacency behavior (combining traces in the same net when they're nearby) is a bit better than thickness, and maybe a good idea to do prior to this

/bounty $150

babyo77, 2025-08-03:
/attempt #66

ainanasolo2005-commits, 2026-02-01:
/attempt #66

CuboYe, 2026-03-14:
Execution refresh from Tez (Mar 14):

- Issue remains OPEN with active bounty and prior territory claim from contributor side: https://github.com/tscircuit/tscircuit-autorouter/issues/66#issuecomment-4025066093
- Existing implementation PR is still open and directly claims this issue: https://github.com/tscircuit/tscircuit-autorouter/pull/638
- PR #638 currently shows green checks (Vercel + benchmark instructions) and benchmark follow-up comments by author.

Given the active overlapping implementation, I am not opening a duplicate competing PR in this cycle to avoid reviewer bandwidth waste.

If maintainers want a second independent implementation/review lane, I can start immediately and post a scoped first-attempt PR in the next cycle.

CuboYe, 2026-03-15:
Post-rp2040 pivot activation: I’m taking a focused next-attempt lane on #66 with a test-first slice.

Execution plan (unit-test aligned):
1) Add a narrow failing regression test covering trace_thickness propagation through the routing path (2x/4x/8x representative cases).
2) Implement minimal fix to ensure thickness parameter reaches route generation consistently.
3) Run targeted tests + full suite subset, then open PR with `/claim #66`.

Decision gate: if maintainers prefer continued consolidation on existing PR #638, I will shift this into a review-assist/test-hardening patchset instead of opening a competing implementation PR.

sayidilxs-web, 2026-05-11:
/attempt #66