tscircuit/tscircuit-autorouter #135: Autorouting Bug Report: <board#148 />

ShiboSoftwareDev, 2025-06-08:
Bug Report Link: https://api.tscircuit.com/autorouting/bug_reports/view?autorouting_bug_report_id=c5e14213-824c-4ce6-9d6d-cb36f4fbc902

this happened after I used the square plating plated hole
[image]

ShiboSoftwareDev, 2025-06-08:
I guess it just needs to be defined as an obstacle

seveibar, 2025-06-09:
Possibly solved in https://github.com/tscircuit/tscircuit-autorouter/pull/139 being released as 0.0.71. Will check shortly using the debugger

seveibar, 2025-06-09:
Confirmed fixed in 0.0.71, thanks for the report @ShiboSoftwareDev !

[image]

closed 2025-06-09 by seveibar