tscircuit/tscircuit-autorouter #235: Autorouting Bug Report: <board#7959 />

Heinrich-XIAO, 2025-09-10:
Bug Report Link: https://api.tscircuit.com/autorouting/bug_reports/view?autorouting_bug_report_id=aa1d41d3-32f4-4a63-80c2-c5d9186d72cf

(any additional details? Maybe a screenshot?)

seveibar, 2025-09-13:
for people looking for the bad trace, it's here

[image]

The DRC check doesn't pick it up, here's the checklist to claim the bounty:

- [ ] Create a reproduction in a test fixture
- [ ] Fix `Debug > Run DRC Checks` to correctly detect the trace that is close to a pad (there must be a minor logic error)

[image]

- [ ] Fix the "multi simplified path solver" to not bring the trace so close

/bounty $400

seveibar, 2025-09-13:
/bounty $400

Heinrich-XIAO, 2025-09-14:
/claim

closed 2025-09-14 by seveibar

Heinrich-XIAO, 2025-09-14:
/attempt #235

PR #242 (2025-09-14): Fix #235 - simplified path solver violating DRC error
Fixed #235 by updating version of `@tscircuit/math-utils` and using correct function to not use manhattan distance.